farmerbriantee/AgOpenGPS_Boards #2: Possible bug in USB autosteer INO

jkonno, 2023-03-06:
if (steerConfig.IsUseY_Axis)
                      **bno08xPitch** = (bno08x.getPitch())* CONST_180_DIVIDED_BY_PI; // Convert pitch to degrees
                    else
                      bno08xRoll = (bno08x.getRoll()) * CONST_180_DIVIDED_BY_PI; //Convert roll to degrees

Line 602 onwards, should probably be

                    if (steerConfig.IsUseY_Axis)
                      **bno08xRoll** = (bno08x.getPitch())* CONST_180_DIVIDED_BY_PI; // Convert pitch to degrees
                    else
                      bno08xRoll = (bno08x.getRoll()) * CONST_180_DIVIDED_BY_PI; //Convert roll to degrees

lansalot, 2023-03-07:
For ref: [https://github.com/farmerbriantee/AgOpenGPS_Boards/blob/master/ArduinoModules/USB/Autosteer_USB_v5/Autosteer_USB_v5.ino][(url](https://github.com/farmerbriantee/AgOpenGPS_Boards/blob/master/ArduinoModules/USB/Autosteer_USB_v5/Autosteer_USB_v5.ino))

farmerbriantee, 2023-03-09:
Redid it so it is like this now, should be better:

                    if (steerConfig.IsUseY_Axis)
                    {
                      bno08xRoll = (bno08x.getPitch())* CONST_180_DIVIDED_BY_PI; // Convert pitch to degrees
                      bno08xPitch = (bno08x.getRoll()) * CONST_180_DIVIDED_BY_PI; //Convert roll to degrees
                    }
                    else
                    {
                      bno08xRoll = (bno08x.getRoll()) * CONST_180_DIVIDED_BY_PI; //Convert roll to degrees
                      bno08xPitch = (bno08x.getPitch()) * CONST_180_DIVIDED_BY_PI; //Convert roll to degrees
                    }

closed 2023-03-09 by farmerbriantee