farmerbriantee/AgOpenGPS_Boards #17: IBT2 schematics is wrong

mirh, 2023-04-16:
https://github.com/farmerbriantee/AgOpenGPS_Boards/blob/master/Misc/Schematics/steer.png
The first big pin next to the capacitor is negative/ground, not +12V. 
I'll grant it's not a walk in the park to figure out when [somebody](https://electropeak.com/learn/wp-content/uploads/2021/01/BTS7960-43A-Driver-Pinout.jpg) say [something](https://www.mpja.com/download/37225MD%20Data.pdf) and others say [otherwise](https://www.handsontec.com/dataspecs/module/BTS7960%20Motor%20Driver.pdf#page=4).. but as can be seen literally in the picture in the last link, there's little wiggle room to misinterpret ***B-*** and ***B+***.

gunicsba, 2023-04-19:
Hello @mirh ,
Would this be a better picture?
https://github.com/gunicsba/AgOpenGPS_Boards/blob/issue17/Misc/Schematics/steer.png

closed 2023-10-07 by jaapvandenhandel

mirh, 2023-10-08:
Cool. 
Btw another small nitpick I just remembered could have been useful to underline, is that (duh) you have a capacitor in the loop. 
In the sense that even when you remove the power, you still are gonna have a remarkably nominal tension for even almost a good minute. Like.. we probably spent an extra half a hour because (in the process of checking whether some spare relays we had were actually working or not I think?) we had misdiagnosed the situation due to the multimeter still not reporting 0. 
A very stupid mistake in hindsight, but this is the kind of rookie mistake that should quite be covered somewhere.