farmerbriantee/AgOpenGPS_Boards #22: Support every IMU orientation

mirh, 2023-04-24:
Or well, every possible one that is aligned with the axes at least. 
[image]
At the moment, with the X/Y toggle I believe these are the only configurations supported (I suppose that if you don't care about the z-axis, "invert roll" may also equal to ones with the board flipped upside down?)

[image] <br> Though it would be nice if these were also accounted for. 
.. yes this is kinda because my BNO is currently mounted in the bottommost left way (because of reasons, not last that outside of [here](https://discourse.agopengps.com/t/imu-orientation-panda/10252) it doesn't seem like anybody took the bother to ever document there was a preferential setup). 
But even though I could fix that without *too much* bother (the biggest annoyance is probably just praying to the god of cables for nothing to come loose at the slightest touch, when you open the box with all the controlling logic inside), I feel like it's backwards to require "flexibility" on the side of the hardware when it's something that could easily be accounted by software. 

p.s. speaking of which, I see the static version of the [SparkFun](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library) library that you ship is 3 years old by now. Maybe you might be interested into updating it? (or maybe you could check the new one by [Adafruit](https://github.com/adafruit/Adafruit_BNO08x), for as much as idk what the pros and cons would be)

gunicsba, 2023-04-24:
By detecting gravity we could figure 1 axis. We still can't auto-detect which of the other 2 is pitch and roll.
https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/commit/611911db6cc62959cf84dac2d7d25a414909c061

mirh, 2023-04-26:
That's pretty neat (and lucky, given it just landed in the last release)

And then maybe the X axis could be calculated too by asking to "accelerate forward" in the wizard?
That way you are only left guessing which direction Y has (which is kinda what "invert roll" already exists for)... if it isn't that you could also handle it automatically by requesting the user to steer left/right while in motion. 
Depending on the accuracy of the sensors (if there [even](https://www.ceva-dsp.com/wp-content/uploads/2019/10/BNO080_085-Datasheet.pdf#page=28) is *one*, idk) all of this might also account for whatever slightly odd angles people could have with their boxes/mountings/fasteners. 

---
That would also call for a revamp of the IMU control panel though (on top of my bemoaning in https://github.com/farmerbriantee/AgOpenGPS/issues/311) to eventually expose the "calibration values" for big guys interested into fine tuning them. 
But aside of the navel gazing over whether stuff couldn't be faster/smoother/abstracter with every adjustment sitting together in the ([software](https://github.com/farmerbriantee/AgOpenGPS/blob/5.7.2/SourceCode/GPS/Forms/UDPComm.Designer.cs#L103)) frontend or ([hardware](https://github.com/farmerbriantee/AgOpenGPS_Boards/blob/master/ArduinoModules/USB/Autosteer_USB_v5/Autosteer_USB_v5.ino#L601)) backend, I'm skeptical about the current "just think to each single axis independently" flow. 

I mean.. it took me two days of [glitches](https://github.com/farmerbriantee/AgOpenGPS/issues/313) to figure out my configuration was actually bust, and that my idiot brain had mixed the pitch and roll axes (because it just so wickedly happened, that when I went to test for roll.. I used a wedge to tilt the tractor, but that also included some noticeable slant backwards which is what I took for good).
I believe the user should be *lead to think* to the whole IMU situation ("which way is my board actually oriented") as opposed to just individual dimensions ("is this one axis inverted/rotated"). 

Like, the later could still be a thing into each specific section of the IMU wizard (as hinted above), but wouldn't it be more intuitive and direct elsewhere if the user could just explicitly input an actual 3D orientation with respect to the vehicle forward plane? How.. I'm not really sure, but the the first dumb idea that came to my mind, is something like a window with each possible state pictured (like the images in my first post) and numbered. But I'm digressing by now I guess.

gunicsba, 2023-04-27:
I think if we could auto discover gravity and set Z accordingly then the user would just need X / Y  and an invert axis button.

As a best practice: don't fix your box until you figure out these roll values.

mirh, 2023-04-27:
My box could have one and only way to be fixed actually, such is the wiggle room I'm afforded in the cabin. 
If any what I could still act upon is its internal side on which to glue the IMU (even though the one it is now was chosen to maximize components spacing, so as I hinted above changing it would give me a worse time with wires management). 

Anyhow, I suppose here the focus should just be on the arduino side of things. 
But first of all, I think all the big guys should discuss over whether you could (or even *should*) have it reason in terms of arbitrary angles, or still keep calling it a day with fixed 90° approximations.

gunicsba, 2023-04-27:
I think we should stick to 90 degrees  (or close to them) actually. (Space is not a concern so I'm sure all of us are able to roughly have these sensors parallel to the tractors axis.)

Here's my thoughts:
We have gravity -> we know 1 axis. Other 2 is unknown.
The same way we're detecting direction we could have an extra loop for the IMU where we'd be driving forward and detect that forward motion so we we have another axis.

Then the 3rd one will be for our roll.

I'm not that good with Arduino but this should totally be feasible.