farmerbriantee/AgOpenGPS_Boards #29: 4.1 STD Boards have TX and RX pins reversed on the BYNAV slot.

chriskinal, 2023-11-11:
The schematic and the PCB files show pin 15 (TX1) on the BYNAV slot goes to pin 29 (TX7) on the Teensy. It also shows pin 16 (RX1) on the BYNAV slot goes to pin 28 (RX7) on the Teensy. I was helping user tomischmid troubleshoot his board. He discovered he had to use Dupont wires to reverse TX/RX  between a UM982 (same pinout for TX1/RX1 as Bynav) to get AOG to see the GPS data. I loaded the schematic and PC files into EasyEDA and confirmed the TX/RX lines are reversed.

m-elias, 2023-11-15:
Any idea if the ByNav also needs CANRX2 and CANTX2 swapped?
[image]

chriskinal, 2023-11-15:
I don't see the CAN connections on the schematic I obtained from the repo a few days ago. AFAIK the UM982 has not implemented CAN yet. The UM982 uses pin 18 for serial TX2 and pin 19 for RX2 just like the Bynav.

m-elias, 2023-11-15:
Don't let the name confuse you, there's no actual canbus involved, that's just the name of the nets on the schematic. It's the other serial port on the bynav, it has two connected to the Teensy. Does it need swapping too or should we just drop support for the bynav as literally noone has even tried it yet.

chriskinal, 2023-11-15:
I have not seen anyone try the Bynav slot except to connect a UM982. Here is a schematic I did to add support for the UM982. I have the EasyEDA as well if you need it.

[image]

m-elias, 2023-11-15:
Does your module have pins 1-4 or 25-28 available? Mine aren't soldered and I can't see any traces to them either nor can I find a connection with a meter.

Do we need both uarts connected? I understand it can send all the data/msgs needed out and receive rtcm on one uart, is that correct?

chriskinal, 2023-11-15:
Currently pins 1 - 4 and 25 - 28 are not populated. On my board pin 25 has a pad but no pin. I did test and the pad is driven high when there is an RTK fix. Adding a pin will not be a problem as far as I can see.

For AOG operation only one UART is needed. Perhaps connecting UM982 UART 2 to a header would facilitate configuration like the USB port on the F9P.

m-elias, 2023-11-15:
What kind of header? FTDI USB to serial pinout? It needs to be 3.3v levels so maybe there's an ESP8266 serial pinout/header standard online? Maybe just a simple 3 pin 0.1" gnd/tx/rx header.

chriskinal, 2023-11-15:
I think a simple 3 pin 0.1" gnd/tx/rx would be fine. I would specify it as a socket in the BOM versus pins. Brushing up against a  LVTTL TX/RX pin with 5v or 12v would not make the UM982 UART happy. Trying to protect the expensive components from the unexpected.